japaric/f3 #9: ITM/`iprint` send data in 32-bit chunks

japaric, 2016-10-05:
Be careful with alignment.

japaric, 2016-10-28:
Note to self: benchmark this with the `time` module

jcsoo, 2016-11-04:
Have you noticed any problems losing output when sending data quickly? I'm running the same code on the F4 and sometimes lose characters.

I have a feeling that it may be dropping it on the ST-Link side, unfortunately.

japaric, 2016-11-04:
> Have you noticed any problems losing output when sending data quickly?

Not yet but I haven't tried high throughput either.

I wouldn't expect it to lose data though. The `iprintln` has a busy wait loop in it to prevent queuing more data if the bus is already busy. I'd expect that to "throttle" the data rate.

> I have a feeling that it may be dropping it on the ST-Link side, unfortunately.

FWIW, the ST-LINK is an STM32F1 device and I doubt it's clocked at a high frequency.

japaric, 2017-08-02:
Done in cortex-m some time ago.

closed 2017-08-02 by japaric