japaric/f3 #44: Async IO

japaric, 2016-10-23:
As a first experiment: `Future::poll` based and interrupt free.

The unresolved question is how to make the microcontroller sleep? I usually use `wfi` when there's nothing to do but that will block forever if no interrupts are being used.

getreu, 2017-01-23:
I have seen with interest your `aio-concurrent,rs` example in your aio branch. You have not followed this branch. I guess you are not satisfied with this approach. Why?

BTW, an event loop with [tokio-core](https://tokio.rs/docs/getting-started/reactor/) looks like this.

``` rust
loop {
    // Learn what the next "event" was, blocking if none available
    let event = next_event();

    // Dispatch this event, following it through to completion
    dispatch(event);
}

Is it possible to do something similar?

Or, can you use `tokio` or `tokio-core` on these devices? When would it make sense?

japaric, 2017-02-27:
@getreu I'm exploring a more "reactive" / "event driven" approach based solely on interrupts but I have nothing to show ATM.

I can't comment on tokio-core as I have not checked it in detail. My understanding is that it doesn't compile on no_std, so there's that.

getreu, 2017-02-28:
@japaric thank you for your answer. I am looking forward to learn more from your publications.

japaric, 2018-01-21:
This crate now has async support via the macros in the [nb](https://docs.rs/nb/0.1.1/nb/) crate. WFI is still a problem. I have an imperfect solution in mind that'll cover in a blog post (but not any time soon).

closed 2018-01-21 by japaric