japaric/f3 #51: Support for naze32 flight controller

kteza1, 2016-10-28:
Hi @japaric. I'm pretty new to rust on microcontrollers. A lot of flight controllers seem to be using STM processors. Does this repo target flight controllers like naze32 too (STM32F103CB) ?

https://www.aliexpress.com/item/NAZE32-REV6-MPU6050-32-bit-6-DOF-10-DOF-Flight-Controller-for-Multicopter-RC-Radio-Control/32649546407.html?spm=2114.30010308.3.94.wRiZWs&ws_ab_test=searchweb0_0,searchweb201602_2,searchweb201603_1&btsid=c352ae82-e4b5-42dc-9332-df00e5bbb71a

Rust for writing flight controller code sounds very exciting to me :)

japaric, 2016-10-28:
> Does this repo target flight controllers like naze32 too (STM32F103CB) ?

Well, not exactly. This crate targets this particular board: The [STM32F3DISCOVERY](http://www.st.com/en/evaluation-tools/stm32f3discovery.html), which has an `STM32F303VCT6` microcontroller in it.

*But* that board, the discovery, does have a gyroscope, a magnetometer and an accelerometer (it's just "9-DOF" as it's missing the baremeter to measure the altitude) and I intend to try my hand at sensor fusion algorithms to convert those raw measurements into the 3D position of the board (something like [this](https://www.youtube.com/watch?v=k4V6apXTnKM)).

All that (sensor drivers, filters, sensor fusion algorithms) shouldn't be too hard to port to other boards but that's only part of what goes into a flight controller. I don't know what else is needed as I haven't write one before. (Although, I do understand the flying mechanism of a quadcopter and have taken control theory courses. Also, I don't own a quadcopter or other aerial unmanned vehicle.)

japaric, 2016-10-29:
@kteza1 I wrote my thoughts on the matter (a flight controller in Rust) [here](https://users.rust-lang.org/t/writing-a-flight-controller-for-drones-in-rust/7340/3?u=japaric).

kteza1, 2016-10-29:
@japaric Wow. That's really awesome. It's a big project but there is a first step for everything. I haven't worked on microcontrollers much but I'll check your `copper` book and try to copy a lot of your `f3` crate to create a crate for `naze32` (as this seems to be the most famous one).

japaric, 2016-10-29:
> create a crate for naze32

Sounds like a great start! Let me know if you need help with anything. I'm on Mozilla's IRC as `japaric`. There's a #rust-embedded channel there that's for this kind of topic.

The copper book needs a small update so if you read it before I get a chance to update it: `rustc` now has built-in targets for ARM Cortex-M processors, like `thumbv7m-none-eabi`; you should use those instead of custom targets defined in `.json` files. Additionally, you'll need to pass some extra flags to the linker; that used to be done with `pre-linker-args` in the `.json` file but with the built-in targets you have to pass those flags via a `.cargo/config` file instead. The docs of the [f3] crate talk about these two things: the built-in targets and the `.cargo/config` file.

[f3]: https://docs.rs/f3/0.2.0/f3/

japaric, 2016-10-29:
Ah, you probably also want to look into code generators for the "register maps". See [svd2rust] and [svd-mmap].

[svd2rust]: https://crates.io/crates/svd2rust
[svd-mmap]: https://github.com/brandonedens/svd-mmap

Oh, and if you are just getting started with microcontrollers, [this] may be of help. It's the workshop material I presented at Rust Belt Rust.

[this]: https://japaric.github.io/rbr2016/

japaric, 2016-10-29:
And I'm going to close this issue since this is not actionable in *this* repo.

closed 2016-10-29 by japaric

kteza1, 2016-10-29:
Sure. Thanks :)