japaric/f3 #71: My board uses STlinkV2, not 2-1

TrionProg, 2016-12-14:
Hello. Great thanks to you, japaric.  Finally I can forget horrible C and  write hight-level code with my favorite programming language. Finally there are direct and safe way how to use this fantastic MCU. Finally I can use one language for all my projects and platforms! Your Discovery book is pretty good, so  detailed and  neat as I prefer. You've done great work.

But where are some problems with your Discovery book:
1) rename pg to f3, I guess =)
2) My board uses STLinkV2, so I used this udev rule
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="3748", \
    MODE:="0666", \
    SYMLINK+="stlinkv2_%n", \
    GROUP:="uucp"
```
STLinkV2's ATTRS{idProduct} must be 3748, not 374b.
Otherwise I have root user of /dev/bus/usb/*/* because this rule do not accept current board.

That is all at the moment.

japaric, 2016-12-14:
Hello @TrionProg. Thanks for the kind words and for the feedback!

> But where are some problems with your Discovery book:

This should have been filled in [the Discovery book issue tracker](https://github.com/japaric/discovery/issues) then ;-).

> rename pg to f3, I guess =)

The pg crate depends on the f3 crate and its job is to limit the functionality/API that f3 exposes. (I feel that incrementally providing more API avoids overloading the reader with too much info). Because of that dependency I can't rename the pg crate; Cargo would yell at me.

> My board uses STLinkV2

Right, there are different hardware revisions of the board so that can happen. I'll add a note about that.

> UDP. Your book has no link to [RM]

It is linked to in one of [the first chapters](https://japaric.github.io/discovery/03-setup/README.html#Documentation) but if you think it should be linked to from somewhere, let me know.

TrionProg, 2016-12-15:
> The pg crate depends on the f3 crate and its job is to limit the functionality/API that f3 exposes. (I feel that incrementally providing more API avoids overloading the reader with too much info). Because of that dependency I can't rename the pg crate; Cargo would yell at me.

Ok. This is good idea. For example, f3 init function should not enable all peripheral. I think,  there should be 3 levels of abstraction(newbie, hight-level library, low-level (registers)).

>It is linked to in one of the first chapters but if you think it should be linked to from somewhere, let me know.

Please add add this links to RTRM chapter. Because I am no full newbie, I have had st-util and other soft installed, so I have missed first chapters =) Next, you give link to discovery manual and later 263 page, etc =)  I had to search manual by google =))

You've made discovery usable, but I need ADC,DMA first, but all this registers are commented
```
// const ADC1: usize = 0x50000000;
// const ADC2: usize = 0x50000100;
// const ADC3: usize = 0x50000400;
// const ADC4: usize = 0x50000500;
```

I want know, why they are not supported yet? I guess, you are developing svd2rust right now. Am I right? As I guess, I should generate memory-map crate from svd. Where I can download svd file?  And can I generate ADC/DMA ? Of course I want use this MCU functions by registers and later to supplement f3 library with hight-level API. My STM32 skill is low yet, but I should already use it to increase my skills.

japaric, 2016-12-16:
> Ok. This is good idea. For example, f3 init function should not enable all peripheral. I think, there should be 3 levels of abstraction(newbie, hight-level library, low-level (registers)).

You can override the default init function by disabling the "default-init" Cargo feature and providing your own init function. [Example](https://docs.rs/f3/0.3.0/f3/examples/override_init/index.html).

> low-level (registers)

The registers can be accessed regardless of where the "default-init" feature is enabled or not.

> Please add add this links to RTRM chapter.

Will do.

> I want know, why they are not supported yet?

Originally, I have been exposing peripherals in an incremental fashion to keep compilation times down. But after #66, it doesn't matter anymore; we can enable all of them right now without affecting the compilation time of the f3 crate.

> I should generate memory-map crate from svd

> And can I generate ADC/DMA ?

You can just extend the [stm32f30x-memory-map](https://github.com/japaric/stm32f30x-memory-map). Just update [this script](https://github.com/japaric/stm32f30x-memory-map/blob/master/gen-peripherals.sh) and re-run it.

> Where I can download svd file? 

There a bunch of SVD files [here](https://github.com/posborne/cmsis-svd/tree/master/data).

> later to supplement f3 library with hight-level API

:+1:. I'd love to see what API you can come up with

japaric, 2018-01-21:
I think there's nothing actionable left in this issue so closing.

closed 2018-01-21 by japaric