japaric/f3 #77: Board support crate naming could be more specific?

Muximize, 2017-06-18:
While short and clear names are nice, pretty and usually the way to go, I have the impression that in the case of board support crates it can cause some confusion, and it might be better to name them after the specific board they support: f3 -> stm32f3discovery

For example, while the device crate stm32f30x is clearly meant for a specific family of devices, the board support crate f3 is used for a single board, but because it's name is less specific it could be interpreted as a device crate for all f3 devices (stm32f3x) or as a board support crate for all Discovery f3 boards (of which there are two now).

There is already an f4 crate as well, which supports the 32f429idiscovery board instead of the more likely stm32f4discovery. What should a crate for that board be called? Or for the other f4 Discovery boards (of which there are six now)?

PS. Thanks for all your hard work towards realizing the dream of Rust on embedded, and especially for doing so in a nice, clean, modular and well documented way! 👍

Emilgardis, 2017-06-18:
I'm currently (very slowly) developing a unpublished f0 crate, and I plan to solve this by using features for when a user needs a specific functionality for a f0 disco/board. Maybe f3 (and f4) could do the same?

japaric, 2017-06-23:
Well, in Rust-land you *always* have to read crate description. For instance, what do you think the crates `clap`, `tokio`, `parking-lot` and `pirate` are used for? 😄

Hopefully once the configurabilty problem is solved the f3 crate won't matter anymore. You'll just grab the stm32f103-hal-impl crate and your configuration file (?) which describes the board hardware and its configuration *for your application* plus some driver crates, `l3gd20` and `lsm303dlhc`, and write your application in a way that it can be compiled for other boards as well. We are not there yet though.

Muximize, 2017-06-23:
Ha yes that's true, but those crates are 'stand-alone' in the sense that they don't have a direct pairing with some other thing that already has a name.

But, if configurabilty is the future, then my point is moot and changing crate names won't be worthwhile for the time left.

closed 2017-06-23 by Muximize