japaric/f3 #81: implement the embedded-hal traits and ...

japaric, 2017-08-02:
make implementations like `Serial` generic around the USART peripheral instance. Basically do the same thing that the [blue-pill] does.

[blue-pill]: https://github.com/japaric/blue-pill

I don't think I'm going to have time to do this myself. So tagging as help-wanted.

cc @brandonedens @Susurrus maybe one of you is interested in tackling this?

tekjar, 2017-08-03:
@japaric I'll try this next week. Will play around and try to understand the framework this week

Susurrus, 2017-08-03:
@tekjar Please CC me on your work for this. I'd be happy to help review it.

japaric, 2017-08-03:
@tekjar :+1:. I believe that @brandonedens was also going to look into this after tackling japaric/svd2rust#107. You might want to let the other know when you start working on this you don't both end up doing the same work.

brandonedens, 2017-09-04:
I added support for the HAL serial at https://github.com/japaric/f3/pull/83
What other HAL interfaces do I need to implement?
@Susurrus @tekjar 
Can you all take a look and confirm that it works on your f3? I'm finding that mine doesn't have proper serial behavior currently nor in the past.

getreu, 2017-09-18:
> Can you all take a look and confirm that it works on your f3?
I can confirm that it compiles in debug and release mode.
Unfortunately I could not try out some examples because I am still struggling with #84.

tekjar, 2017-09-20:
@brandonedens Thanks a lot for this. Build works for me  after adding below lines to `lib.rs`
```
#![feature(const_cell_new)]
#![feature(const_unsafe_cell_new)]
```
Unfortunately `openocd` isn't working on my Mac. Will test on ubuntu in some time.

nordmoen, 2017-10-13:
Is anyone working on SPI support? I have tried to take a stab at it, copying most of the implementation from `blue-pil`, but I'm having some issues when trying to use the implementation together with the built-in `L3GD20` gyroscope.

The trouble I'm having is that the initialization is subtly different between `stm32f30x` and `stm32f103xx` (the later having `gpioa::crl` which I'm unsure how to translate to `stm32f30x`). If anyone is willing to help I would gladly try to implement SPI and support for the gyroscope.

tekjar, 2017-10-13:
@nordmoen I got this working few days back but couldn't structure it well due to lack of time

https://github.com/tekjar/f3/blob/spi/src/spi.rs
https://github.com/tekjar/rust-on-f3/blob/master/demos/l3gd-demo/src/main.rs

nordmoen, 2017-10-13:
Thanks @tekjar I tried with your initialization and now I'm able to communicate! I'll take a stab at creating a `l3gd20` interface, for learning if for nothing else.

nordmoen, 2017-10-16:
So I have tried to implement support for the `l3gd20` gyroscope here: https://github.com/nordmoen/f3/blob/master/src/l3gd20.rs , but I'm having some problems. When trying to read the sensor I always get back `255`. It doesn't matter if it's before or after the sensor is configured. If anyone have a second to look it over I would be very glad for the help.

tekjar, 2017-10-17:
@nordmoen Awesome :). Do you have plans for working on accelerometer as well?

nordmoen, 2017-10-17:
@tekjar Thanks =]. I'll have a look at it, however, since `blue-pill` doesn't have a I2C interface I don't know to structure the implementation, so I will probably wait until someone more talented than me does that.

tekjar, 2017-10-17:
@nordmoen I'll try to work on it after your spi pull request

nordmoen, 2017-10-18:
I still have some questions for the SPI portion of the code. There should be 2 SPI interfaces on the F3Discovery, but the the two `GPIO` modules does not share the same `RegisterBlock` type and so can't be implemented in the same way as with `blue-pill`, is that something that should be solved before a pull-request? In addition, the way my code is structured now it is mostly setup for using `SPI1` for communication with the gyroscope, but it seems it could also be used for external communication, but for that to work it should configure the `NSS` line (?) which is not done in my version as far as I understand.

Unrelated to the above, I was wondering how I can get the gyroscope interrupt to function. I think I'm doing everything correct on the sensor side, but I can't figure out how to set it up within the RTFMv2 framework. I have tried the following to enable interrupt on Line0 selecting `GPIOE` as source(?)
```rust
    // Enable interrupt from Gyroscope
    p.SYSCFG.exticr1.write(|w| unsafe {
        w.exti0().bits(0x04)
    });
    p.EXTI.imr1.write(|w| w.mr0().set_bit());
    p.EXTI.rtsr1.write(|w| w.tr0().set_bit());
```

japaric, 2017-10-19:
Btw, [v0.2](https://github.com/japaric/f3/tree/v0.2.0/src) of this crate had support for both the gyro and the accelerometer. You can look at the code of that release for guidance.

I haven't tried to use the gyroscope interrupt feature or the EXTI* stuff so I can't comment on that.

Also, it's fine to send a PR with gyro API and a single (not generic) SPI API. We can make it generic later.

nordmoen, 2017-10-20:
I have ~stolen almost all~ taken great inspiration from earlier versions of this crate.

I just wanted to test for my self that I understood the interrupt mechanics of the RTFMv2 framework before submitting. The only thing left I think is to properly scale the measurements according to the sensitivity.

japaric, 2018-01-21:
This has been done in v0.5.0

closed 2018-01-21 by japaric