japaric/f3 #82: upgrade to RTFM v0.2

getreu, 2017-08-10:
Hello Jorge,

I like your new API version 2 a lot and I wanted to it try out with my STM32F3discovery board.
It seems that this crate is outdated. Do you plan to upgrade?

phreaknik, 2017-08-30:
@japaric I am in the same boat. I would love to use this `f3` board support crate with the new RTFM v0.2 framework, however the f3 crate does not expose the `Peripherals` struct required by the new RTFM API.

I would fix and submit a PR myself, but I am still learning Rust (using your blog to convert myself from C development to Rust :smile:), and do not yet have the familiarity to propose a solution. Below is the specific error code I get when I try to port the RTFM v0.2 "Hello, world!" program to use the `f3` board:
```
error[E0433]: failed to resolve. Could not find `Peripherals` in `stm32f30x`
  --> src/main.rs:12:1
   |
12 | / app! {
13 | |     // this is a path to a _device_ crate, a crate generated using svd2rust
14 | |     device: f3::stm32f30x,
15 | | }
   | |_^ Could not find `Peripherals` in `stm32f30x`
```

brandonedens, 2017-09-04:
@getreu @phreaknik 
Can you try building the blink example via:
xargo build --example blinky --target thumbv7em-none-eabihf
?
blinky on master uses RTFM v2.

getreu, 2017-09-04:
@japaric it builds! :)
I will try out with the board later.

getreu, 2017-09-04:
Thank you for your help. Here some feedback

1. All examples build without error, e.g.

        xargo build --example blinky --target thumbv7em-none-eabihf
        xargo build --examples  --target thumbv7em-none-eabihf
        xargo build --release --examples  --target thumbv7em-none-eabihf
 
2. The *itm* example did not work for me. Nothing is printed on

        itmdump itm.fifo

3. Somehow I could not set breakpoints

        (gdb) break receive
        Function "receive" not defined.
        Make breakpoint pending on future shared library load? (y or [n]) n

4. I suggest 9600 as default bautrate for the examples. My bluetooth adapter can not do more and
   I suppose this is the case for many other cheap adapters.

5. I could not find an I2C crate. It it pending?

brandonedens, 2017-09-04:
To get a list of available functions try: 
```
(gdb) info functions
```
Do you see that receive is contained in a namespace. Give the full namespace for your breakpoint.

I believe the serial examples were attempting to use the STM32F3 Discovery board version C's USB VCP interface for communication back to the host. When you connected the discovery board to your computer do you notice that a new serial interface is available? On my GNU/Linux machine this is /dev/ttyACM0

We'll have to check in with @japaric about itmdump. I think I was struggling to get those examples up and going myself. I'll try again later and also re-read the discussion of using itmdump from the blog post.

getreu, 2017-09-04:
Thank you! It was a matter of namespace. This works

```
(gdb) info functions
...
(gdb) break preemption::receive
Breakpoint 1 at 0x8001eac: file examples/preemption.rs, line 78.
```

Concerning itm: I found 3 serial devices:

```
$ ls /dev/ttyAC*
/dev/ttyACM0  /dev/ttyACM1  /dev/ttyACM2
```

japaric, 2017-09-04:
@getreu 

> The itm example did not work for me. Nothing is printed on

Contrary to what the DISCOVERY board user manual says solder bridge SB10 is *not* shorted on most (all?) boards so you'll probably have to manually short (e.g. using a solder wire) pin SWO of the SWD header (CN3) to the pin PB3 to get the ITM output passing through the SWD interface then you can use `itmdump itm.fifo`.

Alternatively you can connect PB3 to the RX pin of a UART -> USB adapter, if you have one, and simply "cat" the USB device (e.g. `cat /dev/ttyUSB0`).

getreu, 2017-09-05:
I got it working. My mistake was so simple: I forgot to uncomment the third line in `.gdbinit`. Now it looks like:
```
target remote :3333
monitor tpiu config internal /tmp/itm.fifo uart off 8000000
monitor itm port 0 on
load
```
@japaric I found your article [The ITM and the quest for faster logging](http://blog.japaric.io/itm/) very useful. Thank you for your help! I close this issue and open a new one for I2C support.

closed 2017-09-05 by getreu