japaric/f3 #86: Version 0.5.0 not release to Crates.io

nordmoen, 2017-09-20:
Hi.

I wanted to test some of the examples from this crate. I followed the `cortex-m-quickstart` setup, making sure to follow the `f3` guide removing the correct stuff to get it to work. However, I could not get my code to compile. I see that many dependencies have been updated and the `f3` crate version on github has been updated, but this latest version is not on crates.io.

Is this deliberate or a quick oversight?

japaric, 2017-09-21:
@nordmoen we are still updating things to work with RTFMv2 cf. #85 

Also you can always use the git version:

``` toml
[dependencies]
f3 = { git = "https://github.com/japaric/f3" }
```

> Is this deliberate or a quick oversight?

It's deliberate. A stable release will be released when it's ready to be released.

nordmoen, 2017-09-25:
@japaric Thanks for the quick reply. I hope the sentence did not come a cross as overly negative. I just wanted to ask seeing some of the dependencies was already updated on crates.io.

japaric, 2018-01-21:
v0.5.0 is up now

closed 2018-01-21 by japaric