japaric/f3 #87: Binary size and flashing with st-flash

tekjar, 2018-01-17:
Hi @japaric . The binary size of blinky is around 2.6 mb. Given the flash size is 256K, how is gdb load working? Is there some sort of partial loading?

```
du -sh blinky
2.6M	blinky
```

But debug build fails with `st-flash` while release build works (flashing works, doesn't blink though)
```
st-flash write blinky  0x08000000
```

japaric, 2018-01-17:
> The binary size of blinky is around 2.6 mb. 

That's not the size of the actual binary that will get flashed into the microcontroller. Most of that size are debug symbols. You can get the real size by running `arm-none-eabi-size` on the file.

> st-flash

These flashing utilities usually expect a .bin file; not an ELF, which is the artifact that Cargo produces. You can convert the ELF file into a .bin file using `arm-none-eabi-objcopy` -- I don't recall the exact command off the top of my head though; you'll have to look for it on the internet.

tekjar, 2018-01-17:
Ohh ok. Thanks :)

closed 2018-01-17 by tekjar

tekjar, 2018-01-17:
```
arm-none-eabi-objcopy -O binary  roulette roulette.bin
st-flash write roulette.bin 0x08000000
```

works