japaric/f3 #106: Building fails with: error[E0557]: feature has been removed

Frederik-Baetens, 2019-02-21:
relevant output of trying to run cargo build --example hello:

error[E0557]: feature has been removed
  --> /home/frederik/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-semihosting-0.3.0/src/lib.rs:59:12
   |
59 | #![feature(panic_implementation)]
   |            ^^^^^^^^^^^^^^^^^^^^
   |
note: subsumed by `#[panic_handler]`
  --> /home/frederik/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-semihosting-0.3.0/src/lib.rs:59:12
   |
59 | #![feature(panic_implementation)]
   |            ^^^^^^^^^^^^^^^^^^^^

error[E0658]: The attribute `panic_implementation` is currently unknown to the compiler and may have meaning added to it in the future (see issue #29642)
  --> /home/frederik/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-semihosting-0.3.0/src/lib.rs:71:3
   |
71 | #[panic_implementation]
   |   ^^^^^^^^^^^^^^^^^^^^
   |
   = help: add #![feature(custom_attribute)] to the crate attributes to enable

bobgates, 2019-08-11:
Any news? This caught me as well.

rvlander, 2019-08-12:
I could move two step further. I changed the panic-semihostic dependency to `0.5.2` in `cargo.toml`

But then there is a failure on:
```
error: use of deprecated item 'hal::prelude::_embedded_hal_digital_OutputPin::set_low': Deprecated because the methods cannot return errors. Users should use the traits in digital::v2.
   --> src/led.rs:168:18
    |
168 |         self.pex.set_low()
    |                  ^^^^^^^
    |
note: lint level defined here
   --> src/lib.rs:40:9
    |
40  | #![deny(warnings)]
    |         ^^^^^^^^
    = note: #[deny(deprecated)] implied by #[deny(warnings)]

error: use of deprecated item 'hal::prelude::_embedded_hal_digital_OutputPin::set_high': Deprecated because the methods cannot return errors. Users should use the traits in digital::v2.
   --> src/led.rs:173:18
    |
173 |         self.pex.set_high()
    |                  ^^^^^^^^

error: aborting due to 2 previous errors

error: Could not compile `f3`.
```

You can disable it by commenting offending lines in `led.rs` and by removing the deny warnings directive ine `lib.rs`.

But at this step there is a error with the linker:
```
  = note: rust-lld: error: unknown argument: -Wl,-Tlink.x
          rust-lld: error: unknown argument: -nostartfiles
```

I tested with both stable and nighlty.

rvlander, 2019-08-13:
Got it running by modifying `.cargo/config` to:
```
[target.thumbv7em-none-eabi]
runner = 'arm-none-eabi-gdb'
rustflags = [
  "-C", "link-arg=-Tlink.x",
#  "-C", "link-arg=-Wl,-Tlink.x",
#  "-C", "link-arg=-nostartfiles",
]

[target.thumbv7em-none-eabihf]
runner = 'arm-none-eabi-gdb'
rustflags = [
  "-C", "link-arg=-Tlink.x",
#  "-C", "link-arg=-Wl,-Tlink.x",
#  "-C", "link-arg=-nostartfiles",
]

[build]
target = "thumbv7em-none-eabihf"
```

I also add the following line to`~/.gdbinit`:
```
set auto-load safe-path /
```

This configuraiton might be unsafe.

Also, do not forget to install `arm-none-eabi-gdb`

rvlander, 2019-08-13:
Some other examples still won't compile. I guess this crate needs to be refactored to use the updated version of the hall. Any plans?