japaric/f3 #112: run example: error[E0432]: unresolved import `cortex_m_0_6::peripheral::itm`

bss03arg, 2021-01-26:
Hi,

I am new for rust and when I run the example, I got this error:

cargo run --example hello
    Updating `git://crates.rustcc.com/crates.io-index` index
  Downloaded cobs v0.1.4 (registry `git://crates.rustcc.com/crates.io-index`)
  Downloaded cortex-m-semihosting v0.3.7 (registry `git://crates.rustcc.com/crates.io-index`)
  Downloaded madgwick v0.1.1 (registry `git://crates.rustcc.com/crates.io-index`)
  Downloaded mat v0.2.0 (registry `git://crates.rustcc.com/crates.io-index`)
  Downloaded panic-semihosting v0.5.6 (registry `git://crates.rustcc.com/crates.io-index`)
  Downloaded 5 crates (860.0 KB) in 7.82s
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling nb v1.0.0
   Compiling void v1.0.2
   Compiling cortex-m v0.7.1
   Compiling version_check v0.9.2
   Compiling proc-macro2 v1.0.24
   Compiling vcell v0.1.3
   Compiling bitfield v0.13.2
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.60
   Compiling stable_deref_trait v1.2.0
   Compiling cortex-m v0.6.6
   Compiling cortex-m-rt v0.6.13
   Compiling cortex-m v0.5.10
   Compiling aligned v0.2.0
   Compiling stm32f30x v0.7.1
   Compiling r0 v0.2.2
   Compiling cortex-m-semihosting v0.3.7
   Compiling f3 v0.6.1 (/Users/ccui/CLionProjects/f3)
   Compiling m v0.1.1
   Compiling cobs v0.1.4
   Compiling byteorder v1.4.2
   Compiling volatile-register v0.2.0
   Compiling nb v0.1.3
   Compiling semver v0.9.0
   Compiling generic-array v0.14.4
   Compiling embedded-hal v0.2.4
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling quote v1.0.8
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling generic-array v0.11.1
   Compiling mat v0.2.0
   Compiling l3gd20 v0.2.0
   Compiling madgwick v0.1.1
   Compiling as-slice v0.1.4
   Compiling aligned v0.3.4
   Compiling panic-semihosting v0.5.6
error[E0432]: unresolved import `cortex_m_0_6::peripheral::itm`
  --> /Users/ccui/.cargo/registry/src/crates.rustcc.com-a21e0f92747beca3/cortex-m-0.5.10/src/peripheral/mod.rs:91:46
   |
91 | pub use cortex_m_0_6::peripheral::{cbp, fpb, itm, tpiu};
   |                                              ^^^ no `itm` in `peripheral`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `cortex-m`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed

I use macbook pro with m1 CPU, could you please help me?

Thank you very much!

caemor, 2021-01-26:
You might want to try the new newer board support package here: https://github.com/rubberduck203/stm32f3-discovery which would also fix your issue. 

> error[E0432]: unresolved import cortex_m_0_6::peripheral::itm
--> /Users/ccui/.cargo/registry/src/crates.rustcc.com-a21e0f92747beca3/cortex-m-0.5.10/src/peripheral/mod.rs:91:46
|
91 | pub use cortex_m_0_6::peripheral::{cbp, fpb, itm, tpiu};
| ^^^ no itm in peripheral
>
> error: aborting due to previous error

One of your dependencies seems to rely on cortex-m 0.6 but tries to use the old-api from v 0.5 where itm was still inside peripheral. Currently there seems to be an issue with 0.6.x and 0.5. (see: https://github.com/rust-embedded/discovery/issues/287#issuecomment-767588907)

bss03arg, 2021-01-27:
Thank you very much! I will try the new version of  board support package.

closed 2021-01-27 by bss03arg